APRIL-ZJU/lidar_IMU_calib #1: How to use my own data？

qian5683, 2020-09-01:
Thanks for your good job. But I have some questions about how to use my own data. My lidar device is RoboSense Lidar. and there is a error：

Load dataset from /home/qian/rosbag/lidarimu.bag
terminate called after throwing an instance of 'rosbag::BagFormatException'
  what():  Required 'op' field missing

icameling, 2020-09-03:
@qian5683 

1. Currently, the LI-Calib toolkit only supports VLP-16.
2. If you want to use the other LiDARs, there are two main functions that need to be modified :
 - Add a new lidar model in[ dataset_reader.h]( https://github.com/APRIL-ZJU/lidar_IMU_calib/blob/master/include/utils/dataset_reader.h#L86) and reimplement [unpack_scan](https://github.com/APRIL-ZJU/lidar_IMU_calib/blob/master/include/utils/vlp_common.h#L51)

qian5683, 2020-09-09:
hi~, I've modified the code and used it for my own data. However, there is a problem. The translation I calibrated is very large and does not meet the expectation. I want to know: 
1. What maybe causes this error.
2. Are there any requirements for recording rosbag. For example, what are the requirements for motion trajectory

jayjayhee, 2020-09-09:
Hi,

I think I've modified the two functions you mentioned above properly and the initialization step has been done without errors, But it looks like the toolkit only supports organized pointcloud (currently with a height of 16 using VLP-16), and the lidar I'm using only generates pointcloud data unorganized with a height of 1. So I was wondering if there are any functions I need to modify before I can use this toolkit? I've tried to modify those functions requiring two parameters (width and height) such that they take only one (width), but it doesn't seem to work very well. The data association function returns:

`Plane type  :0 0 0; Plane number: 0
[ WARN] [1599639840.190547592]: [DataAssociation] fails.
`
Please let me know if there's anything I can do to make it work. Thanks!! XXD

Best regards
Jay

noodlles, 2020-09-09:
> hi~, I've modified the code and used it for my own data. However, there is a problem. The translation I calibrated is very large and does not meet the expectation. I want to know:
>
> 1. What maybe [...]

hi,you use the toolkit on your own car(much bigger than the paper)?It can almost only perform plane motion,which cannot ensure imu have sufficient linear acceleration and rotational velocity in x,y,z axis?

icameling, 2020-09-10:
> hi~, I've modified the code and used it for my own data. However, there is a problem. The translation I calibrated is very large and does not meet the expectation. I want to know:
>
> 1. What maybe [...]

@qian5683 

**Data requirements** 

- First, it is important to have the measurement data synchronized between the systems, and we strongly recommend hardware time synchronization. Otherwise, some intrinsic properties of sensor networks such as limited resources of storage, computation, and bandwidth, combined with a potentially high density of nodes make the timestamps of measurements inaccurate.

  Form [the provided datasets](https://github.com/APRIL-ZJU/lidar_IMU_calib#dataset), we can see the timestamps of raw measurements distribute unevenly. 
[image]

  And after synchronization and preprocessing, the timestamps of synchronized measurements arranged evenly.
[image]
（you can use [rqt_bag ](http://wiki.ros.org/rqt_bag) to plot time-series of message）

- Second, based on the observability of IMU-LiDAR extrinsic calibration[1,2], at least two rotations about different axes are necessary. Meanwhile, large accelerations and angular velocities make the estimation more accurate, but you should make sure the error of the LiDAR odometry is within a certain range under that severe motion.

[1] LIC-Fusion 2.0 https://arxiv.org/pdf/2008.07196.pdf (Section IV. OBSERVABILITY ANALYSIS)
[2] Hesch, Joel. (2016). Consistency Analysis and Improvement for Vision-aided Inertial Navigation. Retrieved from the University of Minnesota Digital Conservancy, http://hdl.handle.net/11299/180208. (Section 2.6 IMU-Laser Scanner Extrinsic Calibration)

narutojxl, 2020-09-11:
Hi author, 
I don't have a hardware to do time synchronization, use `rqt_bag` it seems msgs distribute evenly.
[image]

* When `Initialization` finished, the initial value of rotation seems correct. My imu basically aligins with laser(rotation approximate identity) 
> [Initialization] Done. Euler_ItoL initial degree:  3.78477 0.405341 -1.24306
* `Association` shows that we have got `Plane type  :65  1 55; Plane number: 121`.
*  When `Batch` finished,  the result is different from the initial value. It seems thant it converges to the wrong solution.
>`euler_ItoL  :  57.4901  5.51252 -8.14225`
* When `Refine` finished, plane number decrease, not increase.
>Plane type  :16  8 11; Plane number: 25
...
refine 
refine
...

The NDT_target_map.pcd basically meets the environmental scene, but the refined_map.pcd is mess.
[image]
[image]

icameling, 2020-09-15:
@narutojxl 

Please check out the surfels map after the first `Association `, and make sure there are enough surfels in the direction of x-axis, y-axis and z-axis. It's better to collect calibration sequence in a structured environment with big planes

narutojxl, 2020-09-15:
@icameling, thanks for your advise, I will found a place meets this requirements to do experiment :)

narutojxl, 2020-09-16:
@icameling, hi author,
Following your advice, I do the experiment in a small room surrounded by walls. I move imu and laser around and along X,Y, Z axis separately to sample data. After first `Association` finished, there are many planes fit the walls:
[image]
when first `refine` finished, the result is something wrong. The result has not improved, even if `refine` is continuously executed. My param is as following(time offset got from rqt_bag, by manual observation). The bag data is [here,](https://drive.google.com/file/d/1-ce3GweXHWK8WuUpw2GufO74hXSW4QWR/view?usp=sharing), 
>
    <arg name="topic_imu"           default="/imu/data" />
    <arg name="bag_start"           default="1" />
    <arg name="bag_durr"            default="100" />   
    <arg name="scan4map"            default="40" />   
    <arg name="lidar_model"         default="VLP_16" />
    <arg name="ndtResolution"       default="0.5" /> 

    <arg name="time_offset_padding" default="0.001" /> 
    <arg name="show_ui"    default="true" />
The terminal result is:
Note:  Although the terminal display has converged, the result is completely wrong.
>
	Load dataset from /home/jxl/Documents/lidar_imu_calib_result/C1_11F_lidar_imu_sampledata/2020-09-16-15-59-39_big_house.bag
	/velodyne_packets: 910
	/imu/data: 5000
	Framebuffer with requested attributes not available. Using available framebuffer. You may see visual artifacts.

	Ceres Solver Report: Iterations: 51, Initial cost: 6.906546e+05, Final cost: 3.015079e+02, Termination: NO_CONVERGENCE
	[Initialization] Done. Euler_ItoL initial degree:  178.244  -177.28 -172.971

	[Association] start ....
	Plane type  :148 103 180; Plane number: 431
	Surfel point number: 81558
	[Association] 67809.2 ms

	================ Iteration 0 ==================
	Ceres Solver Report: Iterations: 121, Initial cost: 1.197994e+07, Final cost: 5.879937e+05, Termination: CONVERGENCE

	============== After optimization ================
	[Gyro]  Error size, average: 4585;  0.134748  0.133536 0.0563047
	[Accel] Error size, average: 4585;  0.0783432 0.0432497 0.0950025
	[LiDAR] Error size, average: 81558; 0.184476

	P_LinI      :  0.0096321 -0.0583626   -1.15264
	euler_LtoI  :  2.99366 0.706655  58.5238
	P_IinL      : 0.0886249 0.0821533   1.14781
	euler_ItoL  : 177.834 177.816 121.499
	time offset : 0
	gravity     :   -1.8447 -0.335032   9.60879
	acce bias   :  0.250138 -0.192866 0.0372121
	gyro bias   :   0.0103923 -0.00936599 -0.00503104
	[BatchOptimization] 163356 ms

	================ Iteration 1 ==================
	[Association] start ....
	Plane type  :5 1 7; Plane number: 13
	Surfel point number: 1000
	[Association] 18330.9 ms
	Ceres Solver Report: Iterations: 201, Initial cost: 1.993042e+05, Final cost: 3.300348e+04, Termination: NO_CONVERGENCE

	============== After optimization ================
	[Gyro]  Error size, average: 4585; 0.0514282 0.0520165 0.0321376
	[Accel] Error size, average: 4585;  0.0208095 0.0155759 0.0465463
	[LiDAR] Error size, average: 1000; 0.197449

	P_LinI      : -2.32769 -3.49175   2.3051
	euler_LtoI  :   177.47 -163.953 -168.067
	P_IinL      : 2.34987 3.17279 -2.7084
	euler_ItoL  :  5.96376  15.1328 -13.0828
	time offset : 0
	gravity     : -6.19972 -3.13915  6.89589
	acce bias   :  3.84131  1.39964 0.670884
	gyro bias   :  0.0160932 0.00051557 0.00186257
	[Refinement] 44252.6 ms

[... 64 lines cut ...]
	[Refinement] 5458.55 ms

	================ Iteration 5 ==================
	[Association] start ....
	Plane type  : 18  12 142; Plane number: 172
	Surfel point number: 901
	[Association] 19696.2 ms
	Ceres Solver Report: Iterations: 29, Initial cost: 2.328773e+04, Final cost: 2.326717e+04, Termination: CONVERGENCE

	============== After optimization ================
	[Gyro]  Error size, average: 4585; 0.0501242 0.0468216 0.0299117
	[Accel] Error size, average: 4585;  0.0203191 0.0135803 0.0396618
	[LiDAR] Error size, average: 901; 0.0112675

	P_LinI      : -3.22544 -4.09933  1.91412
	euler_LtoI  :  7.43774 -16.6043  9.36238
	P_IinL      : 2.98595 3.37628 -3.2491
	euler_ItoL  :  175.09 162.504  172.48
	time offset : 0
	gravity     : -6.34759 -2.80068  6.90713
	acce bias   :  4.41377  1.51216 0.904278
	gyro bias   :   0.0135546 -0.00086725  0.00291006
	[Refinement] 5377.06 ms

Any help and suggestion will be very appreciated a lot ~

zhangfan950304, 2020-11-19:
Hello, author. The lidar I used is different from vpl-16. The lidar horizontal angle can only take 210 degree point cloud data, and the remaining 150 degree point cloud data is Nan. In this case, can I use this procedure to calibrate?

In addition, since the radar used is not vpl-16, the above two files are modified in the program. If it can be used for calibration, do I need to modify other procedures?

zhangfan950304, 2020-11-19:
Hello, the lidar I used is not vpl-16. The horizontal angle of lidar I used can only take 210 degree point cloud data, and the remaining 150 degree point cloud data is Nan. In this case, can I still use this program to calibrate?

In addition, since the radar used is not vpl-16, the above two files are modified in the program. If it can be used for calibration, do I need to modify other procedures?

zhangfan950304, 2020-11-19:
Hello, the lidar I used is not vpl-16. The horizontal angle of lidar I used can only take 210 degree point cloud data, and the remaining 150 degree point cloud data is Nan. In this case, can I still use this program to calibrate?

In addition, since the radar used is not vpl-16, the above two files are modified in the program.

I also want to ask if it can be used for calibration, do I need to modify other procedures?

zhangfan950304, 2020-11-19:
Hello, the lidar I used is not vpl-16. The horizontal angle of lidar I used can only take 210 degree point cloud data, and the remaining 150 degree point cloud data is Nan. In this case, can I still use this program to calibrate?

In addition, since the radar used is not vpl-16, the two files mentioned above have been modified in the program.

I also want to ask if it can be used for calibration, do I need to modify other procedures?

zhangfan950304, 2020-11-19:
Hello, the lidar I used is not vpl-16. The horizontal angle of lidar I used can only take 210 degree point cloud data, and the remaining 150 degree point cloud data is Nan. In this case, can I still use this program to calibrate?

In addition, since the radar used is not vpl-16, the two files mentioned above have been modified in the program.

I also want to ask if it can be used for calibration, do I need to modify other contents of the program?

goktugyildirim, 2021-03-09:
> Hi,
>
> I think I've modified the two functions you mentioned above properly and the initialization step has been done without errors, But it looks like the toolkit only supports organized pointcloud [...]

I got the same error for Velodyne VLP32C, did you solve this problem?

goktugyildirim, 2021-03-12:
> Hi author,
> I don't have a hardware to do time synchronization, use `rqt_bag` it seems msgs distribute evenly.
> [image]
>
> * When `Initialization` finished, the initial value of rotation seems correct. [...]

could you identify why tool behave like this?

guo-zixuan, 2021-03-29:
@goktugyildirim 
Hello, have you Completed lidar-imu calibration?
I have a try by this work,but I failed.Do you have any suggestions else?

Chrislzy1993, 2021-05-18:
> hi~, I've modified the code and used it for my own data. However, there is a problem. The translation I calibrated is very large and does not meet the expectation. I want to know:
>
> 1. What maybe [...]

@qian5683 , hello, i use plane motion imu data to do this calibration, it fail on init part with a NAN fault happend on ceres solver, i check my imu raw data, there is no nan value, and i am confuse and helpless. did you calib with plane motion data as metion below, and how is the result?

af-doom, 2022-09-19:
@goktugyildirim
Hello, have you Completed lidar-imu calibration?
I have a try by this work,but I failed.Do you have any suggestions else?