APRIL-ZJU/lidar_IMU_calib #2: imu's params  overwritten

narutojxl, 2020-09-04:
Hi author@icameling,  
Thanks for your excellent work! 
In [CalibParamManager ctor](https://github.com/APRIL-ZJU/lidar_IMU_calib/blob/master/include/core/calibration.hpp),  acc, gyro and laser weight used in trajectory_manager.cpp is overwritten by const value.
>
    // fine-tuned parameter
    global_opt_gyro_weight = 28.0;
    global_opt_acce_weight = 18.5;
    global_opt_lidar_weight = 10.0; 

When use my own imu, its price is about dozens of RMB, how to set three params? Thanks a lot!
Best

icameling, 2020-09-07:
Now the output error statistics are the original errors, such as point-to-surfel, the errors between IMU measurements and derivatives of estimated trajectory, they are all unweighted.
If you use different types of IMUs, we suggest to output the weighted error, that is, the normalized error, and check out if the magnitude of the error and the noise amplitude are consistent.

narutojxl, 2020-09-08:
Thanks for your detailed reply :), close it now.

closed 2020-09-08 by narutojxl