APRIL-ZJU/lidar_IMU_calib #3: Using unorganized pointcloud data

jayjayhee, 2020-09-09:
@qian5683 

1. Currently, the LI-Calib toolkit only supports VLP-16.
2. If you want to use the other LiDARs, there are two main functions that need to be modified :
 - Add a new lidar model in[ dataset_reader.h]( https://github.com/APRIL-ZJU/lidar_IMU_calib/blob/master/include/utils/dataset_reader.h#L86) and reimplement [unpack_scan](https://github.com/APRIL-ZJU/lidar_IMU_calib/blob/master/include/utils/vlp_common.h#L51)

_Originally posted by @icameling in https://github.com/APRIL-ZJU/lidar_IMU_calib/issues/1#issuecomment-686312769_

jayjayhee, 2020-09-09:
Hi,

I think I've modified the two functions you mentioned above properly and the initialization step has been done without errors, But it looks like the toolkit only supports organized pointcloud (currently with a height of 16 using VLP-16), and the lidar I'm using only generates pointcloud data unorganized with a height of 1. So I was wondering if there are any functions I need to modify before I can use this toolkit? I've tried to modify those functions requiring two parameters (width and height) such that they take only one (width), but it doesn't seem to work very well. The data association function returns:

`Plane type :0 0 0; Plane number: 0`
`[ WARN] [1599639840.190547592]: [DataAssociation] fails.`

Please let me know if there's anything I can do to make it work. Thanks!! XXD

Best regards
Jay

jayjayhee, 2020-09-10:
Besides, in the initialization step, I've tried to print out the single values under function "EstimateRotation", and the dataset I used gave me the following values:

`cov(1) = 0.037328, cov(2) = 0.027064, cov(3) = 0.022867, cov(4) = -0.000651, cov(5) = 0.000001`

Notice that the odom size was already 270 when giving out the values above but the second single value still seems to be way too small. What might cause this? Are there any requirements when collecting dataset (e.g. move the lidar and imu combination around to make sure there are enough rotation measurements in all three axes?). Please let me know if you need more information. Thanks. I appreciate it.

Jay

icameling, 2020-09-10:
@jayjayhee 

Hello,
> cov(1) = 0.037328, cov(2) = 0.027064, cov(3) = 0.022867, cov(4) = -0.000651, cov(5) = 0.000001
> 
> Plane type :0 0 0; Plane number: 0
> [ WARN] [1599639840.190547592]: [DataAssociation] fails.

1. [DataAssociation ](https://github.com/APRIL-ZJU/lidar_IMU_calib/blob/master/src/ui/calib_ui.cpp#L59)falis due to the failure of [Initialization](https://github.com/APRIL-ZJU/lidar_IMU_calib/blob/master/src/ui/calib_helper.cpp), you can check the [data requirements](https://github.com/APRIL-ZJU/lidar_IMU_calib/issues/1#issuecomment-690298646) and make sure at least two rotations about different axes are provided.
2. The toolkit supports unorganized pointcloud. The only requirement is [the timestamp of each point](https://github.com/APRIL-ZJU/lidar_IMU_calib/blob/master/include/utils/pcl_utils.h#L42) available.

jayjayhee, 2020-09-11:
Hello, 

1. I've tried to rotate the lidar-imu setup to make sure there are enough rotation measurements during data collection and the singular value did become larger. However the second singular value cov(2) never reached 0.25 for whatever I did. Here comes the question, by "two rotations about different axes", do you mean two complete 360 degrees rotations about all three axes or simply changes in any two of the euler's angles (pitch, roll, yaw)? 

2. During data collection, do I need to move around to make sure there's linear velocity and acceleration? Or should I stay at a static point and just rotate the lidar-imu setup.

Pardon for my dumb questions. I appreciate your work and look forward to your reply.

Best regards
Jay

jayjayhee, 2020-09-15:
Hello

I'm using a different type of imu so I was wondering if I need to change the following parameters:

`   global_opt_gyro_weight = 28.0;
    global_opt_acce_weight = 18.5;
    global_opt_lidar_weight = 10.0;`

Thanks for your help. I'm looking forward to your reply.

icameling, 2020-09-15:
> Hello,
>
> 1. I've tried to rotate the lidar-imu setup to make sure there are enough rotation measurements during data collection and the singular value did become larger. However the second singular [...]

Hello @jayjayhee 
1. There is no requirement in the direction of the rotation axis. But the larger the angle between the two rotation axes, the better. I think the observability of the IMU-LiDAR system is similar to the IMU-camera system, therefore you can check section 3.4 of [1] for more details about observability analysis. Here are some corollaries from [1]:

> Corollary 3.1. The matrix described by (3.54) is full rank if the IMU-camera rig is rotated about at least two different axes.
>Corollary 3.2. The system described by (3.28) and (3.30)-(3.34) is observable regardless of the linear motion of the IMU-camera rig.

2. Corollary 3.2.  answers your second question. By the way, we have verified that corollary by real-world data. 

[1] Mirzaei, Faraz M.. (2013). Extrinsic and intrinsic sensor calibration. Retrieved from the University of Minnesota Digital Conservancy, http://hdl.handle.net/11299/162506.

icameling, 2020-09-15:
> Hello
> 
> I'm using a different type of imu so I was wondering if I need to change the following parameters:
> 
> ` global_opt_gyro_weight = 28.0; global_opt_acce_weight = 18.5; global_opt_lidar_weight = 10.0;`
> 
> Thanks for your help. I'm looking forward to your reply.

These parameters are not used in the initialization step.

jayjayhee, 2020-09-15:
> These parameters are not used in the initialization step.

Hi

Thank you for your reply!

I am actually in the refinement step and I noticed that these parameters might affect the optimization results. So I'm trying to find out how these numbers were obtained.

Jay

jayjayhee, 2020-09-17:
Hi!

Thanks for your help. I've successfully collected proper datasets and obtained fair results with your suggestions. However the errors are still relatively larger than those obtained from the given datasets. So I'm thinking there might be some parameters that need to be changed.

1. I'm using my own lidar and imu combo other than the default ones so I was wondering if the following parameters should be changed:
`global_opt_gyro_weight = 28.0; global_opt_acce_weight = 18.5; global_opt_lidar_weight = 10.0;`
if so, are there any specific methods to compute these numbers?

2.The imu I'm using has a data rate of 125 Hz, which is smaller than the given datasets. Is this gonna be an issue when estimating the imu trajectory, which will affect the optimization results accordingly? 

3. How was the gyro bias calculated? I noticed that the gyro bias I obtained is relatively larger and I'm guessing this might be the reason why my results are a little bit off?

Thank you so much for offering this excellent work to the community, I really appreciate it and look forward to your reply.

Best regards,
Jay

icameling, 2020-09-18:
Hello @jayjayhee 

1. These three parameters are manually adjusted. If you really want to find out some instructive methods, you can check this [issue](https://github.com/APRIL-ZJU/lidar_IMU_calib/issues/2#issuecomment-688360052).
2. The frequency of IMU is high enough to constrain the smooth of the trajectory, nothing needs to be modified.  Maybe the number of planes and the magnitude of angular velocity will affect the calibration results.
3. The bias of the gyroscope is considered as constant over the calibration sequence. You can check out  [GyroscopeMeasurement](https://github.com/APRIL-ZJU/Kontiki/blob/master/include/kontiki/measurements/gyroscope_measurement.h#L36) for more details.

jayjayhee, 2020-09-27:
Thanks for your help! I truly appreciate it.

closed 2020-09-27 by jayjayhee

zzcczzcc, 2020-12-26:
hello.I use the vel_16, but I get the error:  Load dataset from /home/z/Downloads/t3.bag
terminate called after throwing an instance of 'pcl::IsNotDenseException'
  what():  : Can't use 2D indexing with a unorganized point cloud

what should I do?

autoperson, 2021-01-04:
Do you slove it

> hello.I use the vel_16, but I get the error: Load dataset from /home/z/Downloads/t3.bag
> terminate called after throwing an instance of 'pcl::IsNotDenseException'
> what(): : Can't use 2D indexing with a unorganized point cloud
> 
> what should I do? Can you help me ?

Do you slove it?