APRIL-ZJU/lidar_IMU_calib #6: is there any reference for vlp_common.h?

petersong1905, 2020-10-09:
thx for your transcendent work!

but i don't konw the following concept, is there any paper or document ?

1. float vert_correction[16] // is it the tan of corresponding scan_mapping_16[]?

2. mVLP16TimeBlock[w][h] = h * 2.304 * 1e-6 + w * 55.296 * 1e-6; // does it work accurately that i can ignored pocket, block....

thanks you so much!

[image]

icameling, 2020-10-12:
- I think your questions could be summarized into one question, that is, how to accurately get the timestamp of each lidar point.

- For Velodyne-16,  you could refer to it's [user manual](https://velodynelidar.com/downloads/) (choose Puck Manual). Here is a snap from chapter 9 of that user manual and it coincides with the code. Note that, the timestamps of Velodyne messages `velodyne_msgs::VelodyneScan::ConstPtr`  have been preprocessed according to the hardware time synchronization flag.

[image]

- Additionally, the second  [unpack_scan](https://github.com/APRIL-ZJU/lidar_IMU_calib/blob/master/include/utils/vlp_common.h#L152) function is for simulation data.

petersong1905, 2020-10-16:
@icameling  thanks for your advices.
 i found full_firing_cycle and single_firing  in following github ,and it works now.
[https://github.com/ros-drivers/velodyne](url)

closed 2020-10-16 by petersong1905