APRIL-ZJU/lidar_IMU_calib #7: Pangolin Build Error

zxl19, 2020-10-12:
Hi @icameling !
Thank you for your team's excellent work! I followed the instructions to build the project, when I came to the following steps:

```shell
# Pangolin
cd lidar_imu_calib_beta
./build_submodules.sh
```
I couldn't find the `lidar_imu_calib_beta` folder. Alternatively, I executed the `./lidar_imu_calib/build_submodules.sh` and encountered the following error:

```shell
/usr/bin/ld: warning: found local symbol '__bss_start' in global part of symbol table in file /usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libGL.so
/usr/bin/ld: warning: found local symbol '_end' in global part of symbol table in file /usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libGL.so
/usr/bin/ld: warning: found local symbol '_edata' in global part of symbol table in file /usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libGL.so
/usr/bin/ld: error: can't create dynamic relocation R_X86_64_TLSLD against symbol: guard variable for pangolin::GlSlUtilities::Instance()::instance in readonly segment; recompile object files with -fPIC
>>> defined in CMakeFiles/pangolin.dir/display/image_view.cpp.o
>>> referenced by image_view.cpp
>>>               CMakeFiles/pangolin.dir/display/image_view.cpp.o:(pangolin::ImageView::Render())

/usr/bin/ld: error: can't create dynamic relocation R_X86_64_DTPOFF32 against symbol: guard variable for pangolin::GlSlUtilities::Instance()::instance in readonly segment; recompile object files with -fPIC
>>> defined in CMakeFiles/pangolin.dir/display/image_view.cpp.o
>>> referenced by image_view.cpp
>>>               CMakeFiles/pangolin.dir/display/image_view.cpp.o:(pangolin::ImageView::Render())

/usr/bin/ld: error: can't create dynamic relocation R_X86_64_TLSLD against symbol: guard variable for pangolin::GlSlUtilities::Instance()::instance in readonly segment; recompile object files with -fPIC
>>> defined in CMakeFiles/pangolin.dir/display/image_view.cpp.o
>>> referenced by image_view.cpp
>>>               CMakeFiles/pangolin.dir/display/image_view.cpp.o:(pangolin::ImageView::Render())

/usr/bin/ld: error: can't create dynamic relocation R_X86_64_DTPOFF32 against symbol: pangolin::GlSlUtilities::Instance()::instance in readonly segment; recompile object files with -fPIC
>>> defined in CMakeFiles/pangolin.dir/display/image_view.cpp.o
>>> referenced by image_view.cpp
>>>               CMakeFiles/pangolin.dir/display/image_view.cpp.o:(pangolin::ImageView::Render())

/usr/bin/ld: error: can't create dynamic relocation R_X86_64_DTPOFF32 against symbol: pangolin::GlSlUtilities::Instance()::instance in readonly segment; recompile object files with -fPIC
>>> defined in CMakeFiles/pangolin.dir/display/image_view.cpp.o
>>> referenced by image_view.cpp
>>>               CMakeFiles/pangolin.dir/display/image_view.cpp.o:(pangolin::ImageView::Render())

/usr/bin/ld: error: can't create dynamic relocation R_X86_64_DTPOFF32 against symbol: pangolin::GlSlUtilities::Instance()::instance in readonly segment; recompile object files with -fPIC
>>> defined in CMakeFiles/pangolin.dir/display/image_view.cpp.o
>>> referenced by image_view.cpp
>>>               CMakeFiles/pangolin.dir/display/image_view.cpp.o:(pangolin::ImageView::Render())

/usr/bin/ld: error: can't create dynamic relocation R_X86_64_TLSLD against symbol: guard variable for pangolin::GlSlUtilities::Instance()::instance in readonly segment; recompile object files with -fPIC
>>> defined in CMakeFiles/pangolin.dir/display/image_view.cpp.o
>>> referenced by image_view.cpp
>>>               CMakeFiles/pangolin.dir/display/image_view.cpp.o:(pangolin::ImageView::Render())

/usr/bin/ld: error: can't create dynamic relocation R_X86_64_DTPOFF32 against symbol: pangolin::GlSlUtilities::Instance()::instance in readonly segment; recompile object files with -fPIC
>>> defined in CMakeFiles/pangolin.dir/display/image_view.cpp.o
>>> referenced by image_view.cpp
>>>               CMakeFiles/pangolin.dir/display/image_view.cpp.o:(pangolin::ImageView::Render())

/usr/bin/ld: error: can't create dynamic relocation R_X86_64_DTPOFF32 against symbol: pangolin::GlSlUtilities::Instance()::instance in readonly segment; recompile object files with -fPIC
>>> defined in CMakeFiles/pangolin.dir/display/image_view.cpp.o
>>> referenced by image_view.cpp
>>>               CMakeFiles/pangolin.dir/display/image_view.cpp.o:(pangolin::ImageView::Render())

/usr/bin/ld: error: can't create dynamic relocation R_X86_64_DTPOFF32 against symbol: pangolin::GlSlUtilities::Instance()::instance in readonly segment; recompile object files with -fPIC
>>> defined in CMakeFiles/pangolin.dir/display/image_view.cpp.o
>>> referenced by image_view.cpp
>>>               CMakeFiles/pangolin.dir/display/image_view.cpp.o:(pangolin::ImageView::Render())

/usr/bin/ld: error: can't create dynamic relocation R_X86_64_DTPOFF32 against symbol: pangolin::GlSlUtilities::Instance()::instance in readonly segment; recompile object files with -fPIC
>>> defined in CMakeFiles/pangolin.dir/display/image_view.cpp.o
>>> referenced by image_view.cpp
>>>               CMakeFiles/pangolin.dir/display/image_view.cpp.o:(pangolin::ImageView::Render())

/usr/bin/ld: error: can't create dynamic relocation R_X86_64_DTPOFF32 against symbol: pangolin::GlSlUtilities::Instance()::instance in readonly segment; recompile object files with -fPIC
[... 32 lines cut ...]
>>>               CMakeFiles/pangolin.dir/display/image_view.cpp.o:(pangolin::ImageView::Render())

/usr/bin/ld: error: can't create dynamic relocation R_X86_64_DTPOFF32 against symbol: pangolin::GlSlUtilities::Instance()::instance in readonly segment; recompile object files with -fPIC
>>> defined in CMakeFiles/pangolin.dir/display/image_view.cpp.o
>>> referenced by image_view.cpp
>>>               CMakeFiles/pangolin.dir/display/image_view.cpp.o:(pangolin::ImageView::Render())

/usr/bin/ld: error: can't create dynamic relocation R_X86_64_DTPOFF32 against symbol: pangolin::GlSlUtilities::Instance()::instance in readonly segment; recompile object files with -fPIC
>>> defined in CMakeFiles/pangolin.dir/display/image_view.cpp.o
>>> referenced by image_view.cpp
>>>               CMakeFiles/pangolin.dir/display/image_view.cpp.o:(pangolin::ImageView::Render())

/usr/bin/ld: error: too many errors emitted, stopping now (use -error-limit=0 to see all errors)
collect2: error: ld returned 1 exit status
src/CMakeFiles/pangolin.dir/build.make:2774: recipe for target 'src/libpangolin.so' failed
make[2]: *** [src/libpangolin.so] Error 1
CMakeFiles/Makefile2:139: recipe for target 'src/CMakeFiles/pangolin.dir/all' failed
make[1]: *** [src/CMakeFiles/pangolin.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```

I browsed to see if there were any solutions to this error but failed. It means so much if you could help me with this.

Best wishes.

icameling, 2020-10-12:
That may be a python linking issue. You could refer to these issues.  May that helps you.

- https://github.com/stevenlovegrove/Pangolin/issues/130
- https://github.com/stevenlovegrove/Pangolin/issues/283

zxl19, 2020-10-13:
Hi @icameling !

Thanks for your reply! I tried the methods in the links you mentioned, but the problem persists. The instructions in the issues are not clear enough so I wondered if I did correctly. It would be wonderful if you could elaborate the instructions.
My case seems more like this one [stevenlovegrove/Pangolin#620](https://github.com/stevenlovegrove/Pangolin/issues/620#issue-710067169).

Best regards.

Chrislzy1993, 2021-04-23:
well, interested question! when i build pangolin by `./build_submodules.sh`command, i get error about 'Python.h' couldn't find。 so i download pangolin from github and build it by myself, it successfully

closed 2022-04-15 by zxl19