APRIL-ZJU/lidar_IMU_calib #10: ceres fails to run well with imu data

petersong1905, 2020-10-17:
I try to use my data with HDL32E and SBG IMU. The only thing i change is data_reader.h,to import data and undistortScan.

Everything is good before TrajectoryManager::trajInitFromSurfel ,include EstimateRotation() and undistortScanInMap with NDT odom pose.I got nice map but not sharp enough.

[image]

And i got 500 surfels,but when i add measurement into ceres by TrajectoryManager::trajInitFromSurfel(),the map will mess up.

[image]

Is it any problem with my imu data?  and how can i solve it?

petersong1905, 2020-10-21:
even i got a point cloud map with 
`Plane type  :961 335 198; Plane number: 1494`
the problem still
[image]

Cracks-ll, 2021-11-12:
> even i got a point cloud map with `Plane type :961 335 198; Plane number: 1494` in first associating , the problem still [image]

I always have a mistake at this step:DataAssociation fails. plane type: 0 0 0;plane number 0.Do you know how to solve this problem？

Cracks-ll, 2021-11-12:
> I try to use my data with HDL32E and SBG Ellipse-A (6 axis). The only thing i change is data import and scan undistortion.
> 
> Everything is good before TrajectoryManager::trajInitFromSurfel ,include EstimateRotation() and undistortScanInMap with NDT odom pose.I got a nice map but not sharp enough.
> 
> [image]
> 
> And i got 500 surfels,but when i added addGyroscopeMeasurementst() or addAccelerometerMeasurement() into ceres by TrajectoryManager::trajInitFromSurfel(),the map would mess up(laser pose was totally calculate by Bspline, addSurfMeasurement is ok). And ceres log like following.
> 
> `WARNING: Logging before InitGoogleLogging() is written to STDERR W20201017 20:57:45.519047 11140 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: CHOLMOD warning: Matrix not positive definite.`
> 
> [image]
> 
> Is it any problem with my imu data? and how can i solve it?

I always have a mistake at this step:DataAssociation fails. plane type: 0 0 0;plane number 0.Do you know how to solve this problem？

qyj1996, 2022-06-15:
> > 我尝试将我的数据与HDL32E和SBG椭圆A（6轴）一起使用。我唯一改变的是数据导入和扫描取消失真。
> > 在 TrajectoryManager：：trajInitFromSurfel 之前，一切都很好，包括 EstimateRotation（） 和 undistortScanInMap with NDT odom pose。我得到了一张漂亮的地图，但不够清晰。
> > [image]
> > 我得到了500个surels，但是当我添加GyroscopeMeasurementst（）或addAccelerometerMeasurement（）到Ceres通过TrajectoryManager：：trajInitFromSurfel（）时，地图会搞砸（激光姿势完全由Bspline计算，addSurfMeasurement是可以的）。谷神星的日志如下。
> > `WARNING: Logging before InitGoogleLogging() is written to STDERR W20201017 20:57:45.519047 11140 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: CHOLMOD warning: Matrix not positive definite.`
> > [image]
> > 我的 imu 数据有问题吗？我该如何解决它？
> 
> 我在此步骤中总是有一个错误：数据关联失败。平面类型：0 0 0;平面编号 0.Do 你知道怎么解决这个问题吗？

> 即使我得到了一个点云图与在第一个关联，问题仍然存在`Plane type :961 335 198; Plane number: 1494` [image]

您好，请问您是如何适配自己的数据集的呢，我尝试了很久，还是不能成功，如果您能指点一二，我将不胜感激！

qyj1996, 2022-06-15:
> > > 我尝试将我的数据与HDL32E和SBG椭圆A（6轴）一起使用。我唯一改变的是数据导入和扫描取消失真。
> > > 在 TrajectoryManager：：trajInitFromSurfel 之前，一切都很好，包括 EstimateRotation（） 和 undistortScanInMap with NDT odom pose。我得到了一张漂亮的地图，但不够清晰。
> > > [image]
> > > 我得到了500个surels，但是当我添加GyroscopeMeasurementst（）或addAccelerometerMeasurement（）到Ceres通过TrajectoryManager：：trajInitFromSurfel（）时，地图会搞砸（激光姿势完全由Bspline计算，addSurfMeasurement是可以的）。谷神星的日志如下。
> > > 
> > > [image]
> > > 我的 imu 数据有问题吗？我该如何解决它？`WARNING: Logging before InitGoogleLogging() is written to STDERR W20201017 20:57:45.519047 11140 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: CHOLMOD warning: Matrix not positive definite.`
> > 
> > 
> > 我在此步骤中总是有一个错误：数据关联失败。平面类型：0 0 0;平面编号 0.Do 你知道怎么解决这个问题吗？
> 
> > 即使我得到了一个点云图与在第一个关联，问题仍然存在`Plane type :961 335 198; Plane number: 1494` [image]
> 
> 您好，请问您是如何适配自己的数据集的呢，我尝试了很久，还是不能成功，如果您能指点一二，我将不胜感激！

您好，请问您是如何适配自己的数据集的呢，我尝试了很久，还是不能成功，如果您能指点一二，我将不胜感激！

> 我在此步骤中总是有一个错误：数据关联失败。平面类型：0 0 0;平面编号 0.Do 你知道怎么解决这个问题吗？

您好，请问您是如何适配自己的数据集的呢，我尝试了很久，还是不能成功，如果您能指点一二，我将不胜感激！

> 我尝试将我的数据与HDL32E和SBG椭圆A（6轴）一起使用。我唯一改变的是数据导入和扫描取消失真。
>
> 在 TrajectoryManager：：trajInitFromSurfel 之前，一切都很好，包括 EstimateRotation（） 和 undistortScanInMap with NDT odom pose。我得到了一张漂亮的地图，但不够清晰。
>
> [image] [...]

您好，请问您是如何适配自己的数据集的呢，我尝试了很久，还是不能成功，如果您能指点一二，我将不胜感激！

qyj1996, 2022-06-15:
> 即使我得到了一个点云图与在第一个关联，问题仍然存在`Plane type :961 335 198; Plane number: 1494` [image]

您好，请问您是如何适配自己的数据集的呢，我尝试了很久，还是不能成功，如果您能指点一二，我将不胜感激！

qyj1996, 2022-06-15:
> > 即使我得到了一个点云图与在第一个关联，问题仍然存在`Plane type :961 335 198; Plane number: 1494`![截图来自 2020-10-21 14-26-05](https://user-images.githubusercontent.com/23067387/96681408-c6c2fb80-13a9-11eb-9eb6-cdf497e8bde7.png)
> 
> 我在此步骤中总是有一个错误：数据关联失败。平面类型：0 0 0;平面编号 0.Do 你知道怎么解决这个问题吗？

您好，请问您是如何适配自己的数据集的呢，我尝试了很久，还是不能成功，如果您能指点一二，我将不胜感激！