APRIL-ZJU/lidar_IMU_calib #11: Is this algorithm suitable for unmanned vehicle platform？

Gutsgwh1997, 2020-11-13:
Initialization failed， because conv (2) is smaller than 0.25. Those are what I got in the initialization phase:
cov()   0.188305   0.187903  0.0202252 0.00665783
cov()   0.220604    0.22034  0.0221111 0.00756393
cov()   0.233234   0.232986  0.0222588 0.00762748
cov()   0.235271   0.235019  0.0224777 0.00770385
cov()   0.235397    0.23508  0.0236421 0.00800224

What's the problem with that? Is it because IMU is not fully motivated?

zhangyuan12121, 2021-05-07:
多点几次初始化

Chrislzy1993, 2021-05-13:
@Gutsgwh1997 @zhangyuan12121 hello
when i use imu data from unamnned car on plane motion, i alse fail in the init part. i get wrong info as following:
```shell
Error in evaluating the ResidualBlock.

There are two possible reasons. Either the CostFunction did not evaluate and fill all    
residual and jacobians that were requested or there was a non-finite value (nan/infinite)
generated during the or jacobian computation. 

Residual Block size: 4 parameter blocks x 1 residuals

For each parameter block, the value of the parameters are printed in the first column   
and the value of the jacobian under the corresponding residual. If a ParameterBlock was 
held constant then the corresponding jacobian is printed as 'Not Computed'. If an entry 
of the Jacobian/residual array was requested but was not written to by user code, it is 
indicated by 'Uninitialized'. This is an error. Residuals or Jacobian values evaluating 
to Inf or NaN is also an error.  

Residuals:                0 

Parameter Block 0, size: 4

           0 |         -nan 
           0 |         -nan 
      0.0001 |         -nan 
           1 |         -nan 

Parameter Block 1, size: 4

           0 |         -nan 
           0 |         -nan 
     2.5e-05 |         -nan 
           1 |         -nan 

Parameter Block 2, size: 4

           0 |         -nan 
           0 |         -nan 
      0.0001 |         -nan 
           1 |         -nan 

Parameter Block 3, size: 4

           0 |         -nan 
           0 |         -nan 
           0 |         -nan 
           1 |         -nan 
```
can anyone show me where i am wrong ?

Gutsgwh1997, 2021-05-19:
> 多点几次初始化

是我IMU激励不够，鼠标点烂了也不行呀