APRIL-ZJU/lidar_IMU_calib #14: Initialization fails.

Lachiven, 2020-11-28:
Ceres Solver Report: Iterations: 14, Initial cost: 3.073979e+02, Final cost: 7.853994e-18, Termination: CONVERGENCE
[ WARN] [1606536377.199574253]: [Initialization] fails.
Ceres Solver Report: Iterations: 2, Initial cost: 7.994445e-18, Final cost: 7.422366e-18, Termination: CONVERGENCE
[ WARN] [1606536390.827585457]: [Initialization] fails.
Ceres Solver Report: Iterations: 2, Initial cost: 7.709791e-18, Final cost: 7.606703e-18, Termination: CONVERGENCE
[ WARN] [1606536418.308922434]: [Initialization] fails.
Ceres Solver Report: Iterations: 2, Initial cost: 7.761621e-18, Final cost: 7.714067e-18, Termination: CONVERGENCE
[ WARN] [1606536496.896175485]: [Initialization] fails.
Ceres Solver Report: Iterations: 2, Initial cost: 7.816852e-18, Final cost: 7.785927e-18, Termination: CONVERGENCE
[ WARN] [1606536581.424251185]: [Initialization] fails.
Ceres Solver Report: Iterations: 2, Initial cost: 7.862062e-18, Final cost: 7.837863e-18, Termination: CONVERGENCE
[ WARN] [1606536670.097768013]: [Initialization] fails.

Lachiven, 2020-11-28:
What happend with initialization fails, how to reslove it

Lachiven, 2020-11-30:
Load dataset from /home/XXX/biaodingshujv.bag
/velodyne_packets: 198
/imu: 3693
Ceres Solver Report: Iterations: 15, Initial cost: 2.048761e+02, Final cost: 3.279987e-16, Termination: CONVERGENCE
Vector4d cov0.0192516
Vector4d cov0.0336664
Vector4d cov0.0432187
Vector4d cov0.0517338
Vector4d cov0.0589967
Vector4d cov0.0652474
Vector4d cov0.0707304
Vector4d cov0.0759678
Vector4d cov0.0808257
Vector4d cov0.0854074
Vector4d cov0.0897549
Vector4d cov0.0915151
Vector4d cov0.0918367
Vector4d cov0.0928698
Vector4d cov0.0970758
Vector4d cov0.101158
Vector4d cov0.105023
[ WARN] [1606699085.790641965]: [Initialization] fails.

What's the requeirment of the date?

goktugyildirim, 2021-03-08:
@Lachiven Did you solve this problem? I get "initialization fails". Do you know the root cause of this problem?

Lachiven, 2021-03-20:
@goktugyildirim Hi, I think it's because "x" "y" "z" "Roll" "Pitch" and "Yaw" in your data is not complete. Or you can see the ori code print some related parameters like "cov" of initialization to see the degree of completion.If your data is fine ,I think you need to Initialization for couple of times.

Chrislzy1993, 2021-04-23:
@goktugyildirim , hi

Chrislzy1993, 2021-05-21:
@goktugyildirim , what kind of data did you use? when i use calib data captured around plane motion, it fails to same error too

zouyajing, 2022-05-06:
> @goktugyildirim , what kind of data did you use? when i use calib data captured around plane motion, it fails to same error too

Same problem due to planar motion here

goktugyildirim, 2022-05-08:
I guess initialization means that monocular camera initialization to generate first 3D points to track in order to estimate camera pose in each time stamp. I don't know how they handle this problem

zouyajing, 2022-05-08:
> 

I guess the initialization here is to do hand-eye calibration, instead of get initial 3D points.  

My problem is caused by planar motion. There is no enough excitation for roll pitch and z. So I manually move my sensors in all the directions with full excitation. Then initialization is finished.

goktugyildirim, 2022-05-08:
Ahaha sorry i deal with cam imu at the same time, i mixed :)