APRIL-ZJU/lidar_IMU_calib #17: when i run with my data(vel_16), there is an error

zzcczzcc, 2020-12-26:
Load dataset from /home/z/Downloads/t3.bag
terminate called after throwing an instance of 'pcl::IsNotDenseException'
  what():  : Can't use 2D indexing with a unorganized point cloud
[li_calib_gui-1] process has died [pid 1492, exit code -6, cmd /home/z/catkin_li_calib/devel/lib/li_calib/li_calib_gui __name:=li_calib_gui __log:=/home/z/.ros/log/ed42ec02-4699-11eb-89f8-80e82cca7b75/li_calib_gui-1.log].
log file: /home/z/.ros/log/ed42ec02-4699-11eb-89f8-80e82cca7b75/li_calib_gui-1*.log

autoperson, 2021-01-04:
I also have this problem

dj-boy, 2021-04-04:
请问这个问题解决了吗，谢谢！

leonshaw06, 2021-08-19:
> Load dataset from /home/z/Downloads/t3.bag
> terminate called after throwing an instance of 'pcl::IsNotDenseException'
> what(): : Can't use 2D indexing with a unorganized point cloud
> [li_calib_gui-1] [...]

When you record the bagfile, you should subscribe to the topic /velodyne_packets instead of /velodyne_points.

xiaobai-1-1, 2022-01-24:
如果要记录/velodyne_points而不是/velodyne_packets，应该怎么解决呢？

chinahuangyong, 2022-03-09:
i also have this question. do you solve?