APRIL-ZJU/lidar_IMU_calib #20: Understanding undistortion of pointcloud

SubMishMar, 2021-01-06:
First of all, thanks for this awesome open source package.

https://github.com/APRIL-ZJU/lidar_IMU_calib/blob/4c75b8d2f60fb22d9b8ee99c7d4e32b53781a787/include/core/scan_undistortion.h#L107-L159

I am trying to understand the cloud un-distortion process referenced above. It looks like, in the code you iterate through the entire point cloud in order to deskew it. Basically, you obtain the time stamp of each point and try to determine the IMU pose associated with it and then using the best estimate of the extrinsic calibration you deskew the point cloud. This is what I understand, please correct me if I am wrong. 

My question is, doesn't it make the system very slow when you iterate through the entire point cloud? For instance, I am using a 128 channel lidar with 200,000 points and when I try to iterate through the entire pointcloud, it really takes a long time. Please let me know what you think about it and suggest me some ways in which I can undistort faster.

Thanks!

SubMishMar, 2021-02-22:
I understand now that you use a VLP-16 sensor which is sparser than OS-128. So I think you are not much affected by the size of the pointcloud.

Chrislzy1993, 2021-05-08:
hi,   if the cloud is organized, and the same column cloud is arrvied at same time, so when deal with a column cloud once a time, not to deal a point once a time. this just my point, Thanks @SubMishMar