APRIL-ZJU/lidar_IMU_calib #30: Using livox mid-70

vincentcheng62, 2021-10-06:
I encounter the following error when I try to calibrate my livox-mid70 with handsfree imu

===========================================================================

ubuntu@ubuntu-desktop:~/ws_livox$ ./src/lidar_IMU_calib/calib.sh
topic_imu:=/handsfree/imu
path_bag:=/home/ubuntu/lidar_imu/test1.bag
ndtResolution:=0.5
==============
... logging to /home/ubuntu/.ros/log/14cbd238-2660-11ec-b544-00044bde7cf1/roslaunch-ubuntu-desktop-8159.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu-desktop:44951/

SUMMARY
========

PARAMETERS
 * /li_calib_gui/LidarModel: livox_mid70
 * /li_calib_gui/bag_durr: 30.0
 * /li_calib_gui/bag_start: 1.0
 * /li_calib_gui/ndtResolution: 0.5
 * /li_calib_gui/path_bag: /home/ubuntu/lida...
 * /li_calib_gui/scan4map: 15.0
 * /li_calib_gui/show_ui: True
 * /li_calib_gui/time_offset_padding: 0.015
 * /li_calib_gui/topic_imu: /handsfree/imu
 * /li_calib_gui/topic_lidar: /livox/lidar
 * /rosdistro: melodic
 * /rosversion: 1.14.11

NODES
  /
    li_calib_gui (li_calib/li_calib_gui)

ROS_MASTER_URI=http://localhost:11311

process[li_calib_gui-1]: started with pid [8178]
[ WARN] [1633502308.162116800]: LiDAR model livox_mid70 not support yet.

Load dataset from /home/ubuntu/lidar_imu/test1.bag
terminate called after throwing an instance of 'pcl::IsNotDenseException'
  what():  : Can't use 2D indexing with a unorganized point cloud
[li_calib_gui-1] process has died [pid 8178, exit code -6, cmd /home/ubuntu/ws_livox/devel/lib/li_calib/li_calib_gui __name:=li_calib_gui __log:=/home/ubuntu/.ros/log/14cbd238-2660-11ec-b544-00044bde7cf1/li_calib_gui-1.log].
log file: /home/ubuntu/.ros/log/14cbd238-2660-11ec-b544-00044bde7cf1/li_calib_gui-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
ubuntu@ubuntu-desktop:~/ws_livox$ 

===========================================================================================

What changes can I make to support the lidar livox mid-70? Thanks!

h-rover, 2021-11-01:
Hello,

Did you get any success with the Mid-70? Any insights will be appreciated. Thanks.

zfc-zfc, 2022-09-28:
> Hello,
> 
> Did you get any success with the Mid-70? Any insights will be appreciated. Thanks.

Maybe you can try LI-Init https://github.com/hku-mars/LiDAR_IMU_Init