APRIL-ZJU/lidar_IMU_calib #35: Can I use lidar_IMU_calib in Ubuntu20.04 ROS noetic?

fengyikawhi, 2022-01-20:


Shelfcol, 2022-04-11:
Of course you can, I learned the magnificent code recently on ubuntu 20.04

You-zeyi, 2022-05-05:
> Of course you can, I learned the magnificent code recently on ubuntu 20.04

Hi, may I ask how long it takes to finish batch optimization of garage-01 on your computer?  I have been running it for almost an hour now and it has not ended.  The CPU usage is about 100%. it seems hopeless for it to end.
[image]

Shelfcol, 2022-05-05:
It consumed almost  three or four hour  to  finish  the calibration.

You-zeyi, 2022-05-05:
> It consumed almost  three or four hour  to  finish  the calibration.
> […](#)

Holy cow! Thanks.