APRIL-ZJU/lidar_IMU_calib #40: 运行时遇到的错误

Gouqtitai, 2022-03-22:
录制rosbag后运行，终端提示如下错误：
Load dataset from /home/go/test.bag
/velodyne_packets: 793
/imu/data: 31998
[li_calib_gui-2] process has died [pid 128526, exit code -11, cmd /home/go/zheda_lidar_imu_cailb/devel/lib/li_calib/li_calib_gui __name:=li_calib_gui __log:=/home/go/.ros/log/ded989d8-a9cf-11ec-a92c-0942836b3eb8/li_calib_gui-2.log].
log file: /home/go/.ros/log/ded989d8-a9cf-11ec-a92c-0942836b3eb8/li_calib_gui-2*.log
不知道出错的原因

kxch928925819, 2022-03-22:
#!/usr/bin/env bash

bag_path="/home/ha/rosbag/li_calib_data"

outdoor_sync_bag_name=(
#"Court-01.bag"
#"Court-02.bag"
#"Court-03.bag"
#"Court-04.bag"
#"Court-05.bag"
)

indoor_sync_bag_name=(
"Garage-01.bag"
#"Garage-02.bag"
#"Garage-03.bag"
#"Garage-04.bag"
#"Garage-05.bag"

看看你的文件名，路径！

Gouqtitai, 2022-03-22:
不是这种问题

Gouqtitai, 2022-03-23:
ubuntu 20.04 ，ROS noetic 版本
原因已经查到，在include-&gt;utils文件夹中的dataset_reader.h文件中:
class LioDataset {
    ......bool read{...}
}
这个类里的read函数你定义为bool型，
执行lio_dataset_temp.read(bag_path_, topic_imu_, topic_lidar, bag_start, bag_durr);语句出错，因为read函数没有返回值。

所以改为void型，就不会出错了。
github上有很多类似问题，可能都是这个原因。

q87956256, 2022-03-24:
> ubuntu 20.04 ，ROS noetic 版本 原因已经查到，在include-&gt;utils文件夹中的dataset_reader.h文件中: class LioDataset {     ......bool read{...} } 这个类里的read函数你定义为bool型， 执行lio_dataset_temp.read(bag_path_, topic_imu_, topic_lidar, bag_start, bag_durr);语句出错，因为read函数没有返回值。 所以改为void型，就不会出错了。 github上有很多类似问题，可能都是这个原因。   
> […](#)

老哥，用了你说的方法瞬间程序就可以运行了，感谢感谢！！！，
我想多问下，我用debug模式提示的是如下：
/usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = const velodyne_msgs::VelodyneScan_<std::allocator<void> >; typename boost::detail::sp_member_access<T>::type = const velodyne_msgs::VelodyneScan_<std::allocator<void> >*]: Assertion `px != 0' failed.
网上查了，这个提示是velodyne_msgs::VelodyneScan的boost指针没有初始化，我看代码里面创建boost指针的时候已经赋值了，所以奇怪的是这个函数的返回类型从bool改成void 与这个boost指针有啥关系啊？

Gouqtitai, 2022-03-24:
> 老哥，用了你说的方法瞬间程序就可以运行了，感谢感谢！！！， 我想多问下，我用debug模式提示的是如下： /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access::type boost::shared_ptr::operator->() const [with T = const velodyne_msgs::VelodyneScan_<std::allocator >; typename boost::detail::sp_member_access::type = const velodyne_msgs::VelodyneScan_<std::allocator >*]: Assertion `px != 0' failed. 网上查了，这个提示是velodyne_msgs::VelodyneScan的boost指针没有初始化，我看代码里面创建boost指针的时候已经赋值了，所以奇怪的是这个函数的返回类型从bool改成void 与这个boost指针有啥关系啊？ 用debug模式也没找到具体原因，只有在这个read函数执行完就出错。 并且，我用ubuntu16.04直接编译运行没有问题，老哥知道这什么原因吗？

兄弟，我能力很有限，那我觉得不要改bool，在read函数末尾加个返回值：return true;这样会不会解决你的问题，我试了，一样可以成功运行，之前出错的原因就是因为这个read函数作者没有给返回值，但能不能解决你的问题就不知道了。-_-

q87956256, 2022-03-24:
> 兄弟，我能力很有限，那我觉得不要改bool，在read函数末尾加个返回值：return true;这样会不会解决你的问题，我试了，一样可以成功运行，之前出错的原因就是因为这个read函数作者没有给返回值，但能不能解决你的问题就不知道了。-_-

好的，我加了return true之后也一样能够运行，谢谢老哥了 O(∩_∩)O

sufjanmystery, 2022-06-11:
> 兄弟，我能力很有限，那我觉得不要改bool，在read函数末尾加个返回值：return true;这样会不会解决你的问题，我试了，一样可以成功运行，之前出错的原因就是因为这个read函数作者没有给返回值，但能不能解决你的问题就不知道了。-_-

你好，我加了return函数，数据也读取了，也重新编译了，但是依然报之前的错误：
Load dataset from /home/panda/catkin_li_calib/src/lidar_IMU_calib/test/Garage-01.bag
/velodyne_packets: 298
/imu1/data_sync: 12001
[li_calib_gui-1] process has died [pid 7400, exit code -11, cmd /home/panda/catkin_li_calib/devel/lib/li_calib/li_calib_gui __name:=li_calib_gui __log:=/home/panda/.ros/log/18fc27d4-e953-11ec-b202-d8f883c5a206/li_calib_gui-1.log].
log file: /home/panda/.ros/log/18fc27d4-e953-11ec-b202-d8f883c5a206/li_calib_gui-1*.log
有点难顶，可以把你编译的源代码发我一份吗？邮箱：1463379112@qq.com

q87956256, 2022-06-21:
> > > 老哥，用了你说的方法瞬间程序就可以运行了，感谢感谢！！！， 我想多问下，我用debug模式提示的是如下： /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access::type boost::shared_ptr::operator->() const [with T = const velodyne_msgs::VelodyneScan_<std::allocator >; typename boost::detail::sp_member_access::type = const velodyne_msgs::VelodyneScan_<std::allocator >*]: Assertion `px != 0' failed. 网上查了，这个提示是velodyne_msgs::VelodyneScan的boost指针没有初始化，我看代码里面创建boost指针的时候已经赋值了，所以奇怪的是这个函数的返回类型从bool改成void 与这个boost指针有啥关系啊？ 用debug模式也没找到具体原因，只有在这个read函数执行完就出错。 并且，我用ubuntu16.04直接编译运行没有问题，老哥知道这什么原因吗？
> > 
> > 
> > 兄弟，我能力很有限，那我觉得不要改bool，在read函数末尾加个返回值：return true;这样会不会解决你的问题，我试了，一样可以成功运行，之前出错的原因就是因为这个read函数作者没有给返回值，但能不能解决你的问题就不知道了。-_-
> 
> @Gouqtitai,你好，我加了return函数，数据也读取了，也重新编译了，但是依然报之前的错误： Load dataset from /home/panda/catkin_li_calib/src/lidar_IMU_calib/test/Garage-01.bag /velodyne_packets: 298 /imu1/data_sync: 12001 [li_calib_gui-1] process has died [pid 7400, exit code -11, cmd /home/panda/catkin_li_calib/devel/lib/li_calib/li_calib_gui __name:=li_calib_gui __log:=/home/panda/.ros/log/18fc27d4-e953-11ec-b202-d8f883c5a206/li_calib_gui-1.log]. log file: /home/panda/.ros/log/18fc27d4-e953-11ec-b202-d8f883c5a206/li_calib_gui-1*.log 有点难顶，可以把你编译的源代码发我一份吗？邮箱：[1463379112@qq.com](mailto:1463379112@qq.com)

好的，已发邮箱了，但是我后来改了些东西，你先看看什么问题吧

kxch928925819, 2022-10-11:
哈哈，你下面那串程序实在超出我的能力了，我是错的太多，所以久错🈶️一点收获,岂敢贻笑大方！