APRIL-ZJU/lidar_IMU_calib #42: 结果求出来啦！如何填写进yaml呢！

kxch928925819, 2022-03-30:
[image]
mapping:
    acc_cov: 0.208587
    gyr_cov: 0.97676
    b_acc_cov: 0.00314634
    b_gyr_cov: 0.0492061
    fov_degree:    180
    det_range:     100.0
    extrinsic_est_en:  false      # true: enable the online estimation of IMU-LiDAR extrinsic,
    extrinsic_T: [ 0.576033,-0.134497,0.19392]
    extrinsic_R: [7.36366,0.434608,-6.43675,
                   0.000291061,2.85E-05,4.55E-05,
                   0.38544,-0.453856,0.00630897]

按照作者的程序，标定结果如下，但是我真的不专业，只不过是个测绘技术人员，现在把这个数据写进yaml一起上传，麻烦您看掌掌眼！指导一下！

zarathustr, 2022-09-23:
这个求出来应该是问题很大的 因为你的R都是不正交的

kxch928925819, 2022-09-24:
这是我做的一个旋转的velodyne。类似geoslam等旋转设备！整了几个月不成功，请指导一下！谢谢🙏

WangWenda98, 2023-04-01:
想问一下，编译完启动gui报错您怎么解决的