APRIL-ZJU/lidar_IMU_calib #50: How to make /imu/data_sync data?

Taeyoung96, 2022-09-02:
@icameling   
Thanks for your great work!  
I have some questions about this tool with the Custom ROS bag file.  

I have only topics named `/velodyne_points` and `imu/data`.  
Is this algorithm works with these topics without `imu/data_sync`?  

If not, How to make `imu/data_sync` topic?  
I'd appreciate your help.  

Thanks,

yinhaosen, 2022-09-26:
I think you need to synchronize lidar and imu data. There are some ways of doing it such as from hardware level or simple synchronization algorithms.

Taeyoung96, 2022-09-26:
@yinhaosen   
Thanks for your reply.  
Is there any recommended simple synchronization algorithm?

yinhaosen, 2022-09-26:
ROS has native support interface google `message_filters::sync`.

closed 2022-09-29 by Taeyoung96