APRIL-ZJU/lidar_IMU_calib #51: terminate called after throwing an instance of 'pcl::IsNotDenseException'   what():  : Can't use 2D indexing with a unorganized point cloud

sysyushuo, 2022-09-26:
运行时候报错，对比了下我这边的topic的消息类型为sensor_msgs/PointCloud2

DUOLAIMI1900, 2023-11-01:
我也遇到了这个问题，这个包需与的点云消息类型为: velodyne_msgs/VelodyneScan，我的点云消息类型是sensor_msgs/PointCloud2

xiaotaozide, 2024-06-26:
请问您解决了吗？我也遇到了同样的问题。

Leaveszy, 2024-06-26:
用不了，不同产品的激光雷达消息类型不一样，只能用相应产品公司提供的算法，或者自己看看有没办法修改算法实现同步。

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2024年06月26日 15:23 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [APRIL-ZJU/lidar_IMU_calib] terminate called after throwing an instance of 'pcl::IsNotDenseException' what(): : Can't use 2D indexing with a unorganized point cloud (Issue #51) |

请问您解决了吗？我也遇到了同样的问题。

xiaotaozide, 2024-06-28:
我复现了这位大佬修改的代码，是可以的。https://blog.csdn.net/weixin_45205745/article/details/129462125