APRIL-ZJU/lidar_IMU_calib #56: error: ‘class pclomp::NormalDistributionsTransform has no member named ‘getTargetCells()'

goat-spike, 2023-02-24:
When i catkin_make i got this error:
error: ‘class pclomp::NormalDistributionsTransform has no member named ‘getTargetCells()'
and i find it points to the ndt_omp.h
then i find that in their https://github.com/APRIL-ZJU/ndt_omp 
the getTargetCell member function is defined as 
const TargetGrid& getTargetCells () const {
      return target_cells_;
    }

i change it as
const TargetGrid& getTargetCells() const {
      return target_cells_;
    }
then catkin_make
the error has solved.

charon-cheung, 2023-11-03:
dude, you should say add this function in `ndt_omp.h`, it had this function in old version

goat-spike, 2023-11-08:
yes, dude, thanks，it had this function in old version.