APRIL-ZJU/lidar_IMU_calib #58: 编译完成后，按照官方文档启动gui报错

WangWenda98, 2023-04-01:
[li_calib_gui-1] process has died [pid 13640, exit code -6, cmd /home/wwd/catkin_li_calib/devel/lib/li_calib/li_calib_gui __name:=li_calib_gui __log:=/home/wwd/.ros/log/3a9cdec4-d05c-11ed-af6e-802bf996be09/li_calib_gui-1.log].
log file: /home/wwd/.ros/log/3a9cdec4-d05c-11ed-af6e-802bf996be09/li_calib_gui-1*.log

NozomiWyane, 2023-06-02:
me too have you solved it

NozomiWyane, 2023-06-05:
> me too have you solved it

yes this is an emergancy issue, I noticed that in **dataset_reader.h**

the function read is type bool, you need to add return true; at the end!!!!!
or you change it to void in both cpp and .h file

WangWenda98, 2023-06-06:
Thank you, I have solved it when I start to run the code
 
 
------------------ Original ------------------
From: "NozomiWyane"; 
Date: 2023年6月5日(星期一) 中午12:53
To: "APRIL-ZJU/lidar_IMU_calib"; 
Cc: "Wenda"; "Author"; 
Subject: Re: [APRIL-ZJU/lidar_IMU_calib] 编译完成后，按照官方文档启动gui报错 (Issue #58)

 

  
me too have you solved it
  
yes this is an emergancy issue, I noticed that in dataset_reader.h
 
the function read is type bool, you need to add return true; at the end!!!!!
 or you change it to void in both cpp and .h file

WangWenda98, 2023-06-06:
When starting to run the code, pay attention to the location of the data, and modify the location of the data and the corresponding parameter file as follows
 
 
------------------ Original ------------------
From: "NozomiWyane"; 
Date: 2023年6月2日(星期五) 上午10:11
To: "APRIL-ZJU/lidar_IMU_calib"; 
Cc: "Wenda"; "Author"; 
Subject: Re: [APRIL-ZJU/lidar_IMU_calib] 编译完成后，按照官方文档启动gui报错 (Issue #58)

 

 
me too have you solved it