APRIL-ZJU/lidar_IMU_calib #72: 初始化出错

yuhan0328-s, 2024-12-02:
我在录制完数据包，想要做外参标定的时候。出现如下问题，以前用过的数据包没问题，但是我新录得数据包都会出现这个情况。已知数据包格式没有问题，对比了以前成功的包，而且步骤完全一样。
/velodyne_packets: 991
/imu/data: 10000
Ceres Solver Report: Iterations: 2, Initial cost: 3.920000e-06, Final cost: 3.964010e-14, Termination: CONVERGENCE
[ WARN] [1733140646.770074881]: [Initialization] fails.
请各位看到的话可以帮我解惑，非常感谢！！！

zrxwz, 2024-12-05:
> 我在录制完数据包，想要做外参标定的时候。出现如下问题，以前用过的数据包没问题，但是我新录得数据包都会出现这个情况。已知数据包格式没有问题，对比了以前成功的包，而且步骤完全一样。 /velodyne_packets: 991 /imu/data: 10000 Ceres Solver Report: Iterations: 2, Initial cost: 3.920000e-06, Final [...]

你好，请问你解决了吗？我也遇到了一模一样的问题

CHAIJZ888, 2024-12-10:
你好，我遇到了同样的问题，请问解决了吗？

zrxwz, 2024-12-10:
> 你好，我遇到了同样的问题，请问解决了吗？
把launch文件中显示ui界面改成true，然后在界面中多点几次初始化按钮，我是按了五六次以后才成功，不过最后得到的结果还是很差

zrxwz, 2024-12-10:
> 你好，我遇到了同样的问题，请问解决了吗？

我最近在做azure kinect 和rslidar的标定，彩色相机与雷达，imu和雷达都做过了，但是效果都很差结果没法用，请问你有没有相关经验

CHAIJZ888, 2024-12-11:
我也尝试了几种方法，最后效果都比较差，另外我不依赖ROS，因此需要先保存数据，随后手动写入rosbag来实现。用了仿真中的数据，发现结果差的有点大，我们可以一起交流一下。

zrxwz, 2024-12-11:
> 我也尝试了几种方法，最后效果都比较差，另外我不依赖ROS，因此需要先保存数据，随后手动写入rosbag来实现。用了仿真中的数据，发现结果差的有点大，我们可以一起交流一下。

我今天刚刚完成了一个kinect的rgb相机和rslidar的标定，我们可以加一个联系方式交流。QQ2622916338