APRIL-ZJU/lidar_IMU_calib #75: [li_calib_gui_2]process has died

liTAo0713, 2025-04-14:
i have modified the codes according to the issue 58,but it's not solved yet,anyone know the reason？
[li_calib_gui-2] process has died [pid 18840, exit code -11, cmd /home/litao/catkin_li_calib/devel/lib/li_calib/li_calib_gui __name:=li_calib_gui __log:=/home/litao/.ros/log/45d0f6fa-18f7-11f0-8ad3-8f4a1eab39e1/li_calib_gui-2.log].
log file: /home/litao/.ros/log/45d0f6fa-18f7-11f0-8ad3-8f4a1eab39e1/li_calib_gui-2*.log

ZZ-SHU, 2025-04-16:
I have also encountered this problem and am seeking a solution

ZZ-SHU, 2025-04-16:
bool read(const std::string path,
            const std::string imu_topic,
            const std::string lidar_topic,
            const double bag_start = -1.0,
            const double bag_durr = -1.0) {

    data_.reset(new LioDataset(lidar_model_));
    data_->bag_.reset(new rosbag::Bag);
    data_->bag_->open(path, rosbag::bagmode::Read);

    init();

    rosbag::View view;
    {
      std::vector<std::string> topics;
      topics.push_back(imu_topic);
      topics.push_back(lidar_topic);

      rosbag::View view_full;
      view_full.addQuery(*data_->bag_);
      ros::Time time_init = view_full.getBeginTime();
      time_init += ros::Duration(bag_start);
      ros::Time time_finish = (bag_durr < 0)?
                              view_full.getEndTime() : time_init + ros::Duration(bag_durr);
      view.addQuery(*data_->bag_, rosbag::TopicQuery(topics), time_init, time_finish);
    }

    for (rosbag::MessageInstance const m : view) {
      const std::string &topic = m.getTopic();

      if (lidar_topic == topic) {
        TPointCloud pointcloud;
        double timestamp = 0;

        if (m.getDataType() == std::string("velodyne_msgs/VelodyneScan")) {
          velodyne_msgs::VelodyneScan::ConstPtr vlp_msg =
                  m.instantiate<velodyne_msgs::VelodyneScan>();
          timestamp = vlp_msg->header.stamp.toSec();
          p_LidarConvert_->unpack_scan(vlp_msg, pointcloud);
        }
        if (m.getDataType() == std::string("sensor_msgs/PointCloud2")) {
          sensor_msgs::PointCloud2::ConstPtr scan_msg =
                  m.instantiate<sensor_msgs::PointCloud2>();
          timestamp = scan_msg->header.stamp.toSec();
          p_LidarConvert_->unpack_scan(scan_msg, pointcloud);
        }

        data_->scan_data_.emplace_back(pointcloud);
        data_->scan_timestamps_.emplace_back(timestamp);
      }

      if (imu_topic == topic) {
        sensor_msgs::ImuConstPtr imu_msg = m.instantiate<sensor_msgs::Imu>();
        double time = imu_msg->header.stamp.toSec();

        data_->imu_data_.emplace_back();
        data_->imu_data_.back().timestamp = time;
        data_->imu_data_.back().accel = Eigen::Vector3d(
                imu_msg->linear_acceleration.x, imu_msg->linear_acceleration.y,
                imu_msg->linear_acceleration.z);
        data_->imu_data_.back().gyro = Eigen::Vector3d(
                imu_msg->angular_velocity.x, imu_msg->angular_velocity.y,
                imu_msg->angular_velocity.z);
      }
    }

    std::cout << lidar_topic << ": " << data_->scan_data_.size() << std::endl;
    std::cout << imu_topic << ": " << data_->imu_data_.size() << std::endl;

   return true;
  }

liTAo0713, 2025-04-17:
did you solve it？i compiled it again and now it has another problem again. terminate called after throwing an instance of 'std::out_of_range' what():vector::_M_range_check:__n(which is 78750)＞=this-＞size()(which is 78750)